espressif/esp-who #252: 运行human_face_recognition/lcd报错 serial.serialutil.SerialException

junnan-xjtu, 2023-04-07:
前段时间可以烧录例程，今天发现不止这个例程烧录不了，其他例程也烧录不了了，况且烧录过程中会发出USB设备弹出的提示音，然后就报以下错误，请问是什么原因呢？

Traceback (most recent call last):
  File "D:/ESP32-IDF/esp-tools-offline/Espressif/frameworks/esp-idf-v4.4.4/components/esptool_py/esptool/esptool.py", line 493, in command
    p = self.read()
  File "D:/ESP32-IDF/esp-tools-offline/Espressif/frameworks/esp-idf-v4.4.4/components/esptool_py/esptool/esptool.py", line 438, in read
    return next(self._slip_reader)
  File "D:/ESP32-IDF/esp-tools-offline/Espressif/frameworks/esp-idf-v4.4.4/components/esptool_py/esptool/esptool.py", line 3661, in slip_reader
    raise FatalError(msg)
__main__.FatalError: Serial data stream stopped: Possible serial noise or corruption.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "D:/ESP32-IDF/esp-tools-offline/Espressif/frameworks/esp-idf-v4.4.4/components/esptool_py/esptool/esptool.py", line 5482, in <module>
    _main()
  File "D:/ESP32-IDF/esp-tools-offline/Espressif/frameworks/esp-idf-v4.4.4/components/esptool_py/esptool/esptool.py", line 5475, in _main
    main()
  File "D:/ESP32-IDF/esp-tools-offline/Espressif/frameworks/esp-idf-v4.4.4/components/esptool_py/esptool/esptool.py", line 4907, in main
    operation_func(esp, args)
  File "D:/ESP32-IDF/esp-tools-offline/Espressif/frameworks/esp-idf-v4.4.4/components/esptool_py/esptool/esptool.py", line 4082, in write_flash
    blocks = esp.flash_begin(uncsize, address, begin_rom_encrypted=encrypted)
  File "D:/ESP32-IDF/esp-tools-offline/Espressif/frameworks/esp-idf-v4.4.4/components/esptool_py/esptool/esptool.py", line 818, in flash_begin
    self.check_command("enter Flash download mode", self.ESP_FLASH_BEGIN,
  File "D:/ESP32-IDF/esp-tools-offline/Espressif/frameworks/esp-idf-v4.4.4/components/esptool_py/esptool/esptool.py", line 520, in check_command
    val, data = self.command(op, data, chk, timeout=timeout)
  File "D:/ESP32-IDF/esp-tools-offline/Espressif/frameworks/esp-idf-v4.4.4/components/esptool_py/esptool/esptool.py", line 509, in command
    self._port.timeout = saved_timeout
  File "D:\ESP32-IDF\esp-tools-offline\Espressif\python_env\idf4.4_py3.8_env\lib\site-packages\serial\serialutil.py", line 372, in timeout
    self._reconfigure_port()
  File "D:\ESP32-IDF\esp-tools-offline\Espressif\python_env\idf4.4_py3.8_env\lib\site-packages\serial\serialwin32.py", line 222, in _reconfigure_port
    raise SerialException(
serial.serialutil.SerialException: Cannot configure port, something went wrong. Original message: PermissionError(13, ' 设备不识别此命令。', None, 22)
CMake Error at run_serial_tool.cmake:56 (message):

  D:/ESP32-IDF/esp-tools-offline/Espressif/python_env/idf4.4_py3.8_env/Scripts/python.exe
  D:/ESP32-IDF/esp-tools-offline/Espressif/frameworks/esp-idf-v4.4.4/components/esptool_py/esptool/esptool.py
  --chip esp32s3 failed

Auroragan, 2023-04-10:
Hi,
试一下这个操作：https://github.com/espressif/esp-who/blob/master/docs/zh_CN/get-started/ESP32-S3-EYE_Getting_Started_Guide.md#42-%E5%B7%B2%E7%9F%A5%E9%97%AE%E9%A2%98

junnan-xjtu, 2023-04-13:
谢谢，确实是这个问题，已解决。

closed 2023-04-13 by junnan-xjtu