espressif/esp-who #254: human_face_recognition例程中录入的人脸无法储存

CaoJu600, 2023-04-13:
人脸可以正常录入、对比、删除，但似乎没有储存到flash，重启后仍显示
I (1102) MFN: fr partition size: 131072 bytes, maxminum 62 IDs can be stored
I (1102) MFN: No face ID in flash
lcd和terminal例程都尝试过，芯片为esp32s3 n16r8，例程代码没有任何修改。
不清楚是哪里的问题，希望能帮忙找下原因，谢谢！

下面是重启前到重启后的完整记录：
I (5282) detection_result: [ 0]: ( 50,  55, 172, 206)
I (5282) detection_result:       left eye: ( 82, 107), right eye: (134, 109), nose: (105, 128), mouth left: ( 84, 156), mouth right: (121, 159)
E (5282) RECOGNIZE: Similarity: -1.000000, Match ID: -1
W (7022) ENROLL: ID 1 is enrolled
I (9612) detection_result: [ 0]: ( 58,  58, 167, 196)
I (9612) detection_result:       left eye: ( 84, 105), right eye: (132, 105), nose: (104, 125), mouth left: ( 89, 157), mouth right: (124, 157)
I (9612) RECOGNIZE: Similarity: 0.865758, Match ID: 1
ClearCommError failed (PermissionError(13, '???????????????????????????', None, 22))
Waiting for the device to reconnect.
I (375) esp_image: segment 1: paddr=001b8384 vaddr=3fc961I (902) spiram: SPI SRAM memory test OK
I (911) cpu_start: Pro cpu start user code
I (911) cpu_start: cpu freq: 240000000
I (912) cpu_start: Application information:
I (912) cpu_start: Project name:     human_face_recognition_lcd
I (914) cpu_start: App version:      v4.4.4-dirty
I (915) cpu_start: Compile time:     Apr 13 2023 14:54:26
I (917) cpu_start: ELF file SHA256:  c368d27962e7767d...
I (918) cpu_start: ESP-IDF:          v4.4.4-dirty
I (919) heap_init: Initializing. RAM available for dynamic allocation:
I (921) heap_init: At 3FC9E5A0 len 0004B170 (300 KiB): D/IRAM
I (923) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM
I (924) heap_init: At 600FE000 len 00002000 (8 KiB): RTCRAM
I (926) spiram: Adding pool of 8192K of external SPI memory to heap allocator
W (928) spi_flash: Detected size(16384k) larger than the size in the binary image header(8192k). Using the size in the binary image header.
I (938) sleep: Configure to isolate all GPIO pins in sleep state
I (939) sleep: Enable automatic switching of GPIO sleep configuration
I (0) cpu_start: Starting scheduler on APP CPU.
I (942) s3 ll_cam: DMA Channel=4
I (942) cam_hal: cam init ok
I (942) sccb: pin_sda 4 pin_scl 5
I (942) sccb: sccb_i2c_port=1

I (952) camera: Detected camera at address=0x30
I (952) camera: Detected OV2640 camera
I (952) camera: Camera PID=0x26 VER=0x42 MIDL=0x7f MIDH=0xa2
I (1022) s3 ll_cam: node_size: 3840, nodes_per_line: 1, lines_per_node: 8
I (1022) s3 ll_cam: dma_half_buffer_min:  3840, dma_half_buffer: 11520, lines_per_half_buffer: 24, dma_buffer_size: 23040
I (1032) cam_hal: buffer_size: 23040, half_buffer_size: 11520, node_buffer_size: 3840, node_cnt: 6, total_cnt: 10
I (1032) cam_hal: Allocating 115200 Byte frame buffer in PSRAM
I (1032) cam_hal: Allocating 115200 Byte frame buffer in PSRAM
I (1032) cam_hal: cam config ok
I (1032) ov2640: Set PLL: clk_2x: 1, clk_div: 3, pclk_auto: 1, pclk_div: 8
I (1102) MFN: fr partition size: 131072 bytes, maxminum 62 IDs can be stored
I (1102) MFN: No face ID in flash
I (1102) spi_bus: SPI2 bus created
I (1102) spi_bus: SPI2 bus device added, CS=14 Mode=0 Speed=40000000
I (1312) lcd st7789: MADCTL=0
I (1312) who_lcd: Screen name:ST7789 | width:240 | height:240

vahidrezaee, 2023-09-30:
Hi,
I have the same problem
when I build and flash the human_face_recognition  example on the Esp32S3-eye board.
the registered faces will be erased after restarting the board.

Sandra-lol, 2023-10-25:
Hi guys, the master and idf5.0 branches have been updated, pls try again to see if the issue still exists. Thanks!

MichaelDu9226, 2023-12-13:
idf 4.4 commit:b059440b83c7c3e46a3ed7bb0701a00602ad947d
also have the same problem

> 在全片擦除后注册的第一个人脸可以正常储存，重启后仍能读取到。
> 但如果继续注册第二个人脸或者删除人脸重新注册，则所有注册的人脸(包括第一个)都无法储存，重启就会丢失，必须要再次擦除flash才能恢复。

MichaelDu9226, 2023-12-13:
> Hi guys, the master and idf5.0 branches have been updated, pls try again to see if the issue still exists. Thanks!

I have confirmed that IDF 5.0 can be saved.
However, as our project is based on 4.4, can you please clarify which commit is related to this issue and whether we can resolve it by updating this commit on the v4.4 branch?
我已经确定idf 5.0可以保存。
但是因为我们的工程是基于4.4，请问这个问题有哪个commit有关，我们是否可以通过更新这个commit在v4.4分支解决这个问题。

MichaelDu9226, 2023-12-13:
> Hi guys, the master and idf5.0 branches have been updated, pls try again to see if the issue still exists. Thanks!

@Sandra-lol 
I have confirmed that IDF 5.0 can be saved.
However, as our project is based on 4.4, can you please clarify which commit is related to this issue and whether we can resolve it by updating this commit on the v4.4 branch?
我已经确定idf 5.0可以保存。
但是因为我们的工程是基于4.4，请问这个问题有哪个commit有关，我们是否可以通过更新这个commit在v4.4分支解决这个问题。