espressif/esp-who #256: Dimensions for ESP-EYE mainboard front/back

ansonl, 2023-04-26:
Hi, are there published dimensions for the ESP-EYE mainboard front/back? 

The ESP32-S3-EYE has dimension links at https://github.com/espressif/esp-who/blob/master/docs/en/get-started/ESP32-S3-EYE_Getting_Started_Guide.md but the similar page for ESP-EYE has no dimensions https://github.com/espressif/esp-who/blob/master/docs/en/get-started/ESP-EYE_Getting_Started_Guide.md. 

The ESP-EYE is this board that is smaller and does not have a screen built in like the ESP32-S3-EYE.
[image]

Sandra-lol, 2023-05-05:
Hi, we've created task and will update doc soon.

ansonl, 2023-05-25:
Hi, @Sandra-lol has there been any update on getting the physical spec docs?

Daiziyan, 2023-06-01:
Hi @ansonl , please see the attachment for the dimensions of the ESP-EYE mainboard. Relevant documents will be updated later.
[Dimension_ESP-EYE_V2.2.zip](https://github.com/espressif/esp-who/files/11624880/Dimension_ESP-EYE_V2.2.zip)

ansonl, 2023-06-01:
@Daiziyan Thank you for the update. Will the updated documents show measurements for the single mounting hole on the ESP-EYE? I viewed the attachment and it only had the board's exterior dimension without positioning info of components and sensors on it.

Daiziyan, 2023-06-02:
Hi @ansonl, you can view the .dxf file in the attachment with [Autodesk Viewer](https://viewer.autodesk.com/) online for detailed info of components and sensors on the ESP-EYE board.

ansonl, 2023-07-17:
@Daiziyan Thank you for the file and viewer link. It didn't show the thickness of the parts, but I was able to measure an actual ESP-EYE for that. 
I used it to create a [ESP-EYE case and mount](https://www.printables.com/model/520941-esp-eye-corner-camera-mount) since I didn't find any existing ones available.

closed 2023-07-17 by ansonl