espressif/esp-who #272: How to load a model from an external sd card

Liuyuehuixue, 2023-09-04:
Hi, I have a problem, esp-sr can load models from external sd card. 
https://docs.espressif.com/projects/esp-sr/zh_CN/latest/esp32s3/flash_model/README.html

I got four bin files when compiling speech recognition, and got three bin files when compiling examples such as face recognition. Compared with speech recognition, model.bin is missing, is it merged? May I ask if the example in esp-who can load the model from an external sd card to reduce the flash usage.

PrathamG, 2023-11-07:
Hi! Also interested to know if face recognition model can be used from sd card instead of keeping it in flash

sun-xiangyu, 2023-12-13:
Yes, face recognition model can return the face embedding (currently have not supported this API) and then you can save it anywhere.

MichaelDu9226, 2023-12-28:
@sun-xiangyu When will the function of saving and loading face recognition models from SD cards be completed?

Or how can I modify the version of idf4.4 to ensure that the face model saved in flash can be reloaded after restarting.
Currently, our project is based on idf4.4, but it may not be possible to migrate to idf5.0 in the short term.

Relate this issue:
https://github.com/espressif/esp-who/issues/254