espressif/esp-who #276: ESP32-S3-EYE工程logo替换遇到的问题

FitzzzzZZZZ, 2023-10-08:
我直接将工程ESP32-S3-EYE中AppLCD接口中的logo_en_240x240_lcd数组中的数据替换，显示的颜色异常，形状是对的。
我是将图片通过RGB888转成RGB565格式的16位无符号整形数组，显示的图片如下。
[image]
[image]

FitzzzzZZZZ, 2023-10-08:
[image]
转换前的图片

FitzzzzZZZZ, 2023-10-08:
RGB888转RGB565的程序如下：
from PIL import Image

def image_to_rgb565_array(image_path):
    img = Image.open(image_path)
    img = img.convert('RGB')  # Convert image to RGB mode
    
    width, height = img.size
    rgb565_array = []

    for y in range(height):
        for x in range(width):
            r, g, b = img.getpixel((x, y))
            # Convert RGB to RGB565
            rgb565_value = ((r & 0xF8) << 8) | ((g & 0xFC) << 3) | (b >> 3)
            rgb565_array.append(rgb565_value)

    return rgb565_array

# Replace 'your_image_path.jpg' with the actual path to your image
image_path = 'your_image_path.jpg'
rgb565_array = image_to_rgb565_array(image_path)

# Print the resulting array
print(rgb565_array)

WangYuxin-esp, 2023-10-09:
图像正确、颜色不对的问题，可能是 R\G\B 三分量的顺序与 LCD 要求的 R\G\B 分量的顺序不同导致的。请检查 LCD、图像数据的 R\G\B 分量的顺序是否一致。