espressif/esp-who #282: 颜色识别阈值是什么格式的

Zhengqinze05, 2023-12-20:
我使用了esp-eye开发板，使用了颜色检测的案例，请问这个颜色识别的阈值是什么格式的，我以为他是RGB的，但是看起来不是的，变化最大的是第一个和第二个阈值参数，甚至第二个阈值参数可以比第一个小例如官方的{{156, 10, 70, 255, 90, 255}, 64, "red"}。我很疑惑这个颜色的阈值是什么格式的，RGB?HVS?LAB?

Sean-WangXH, 2024-08-17:
请问您可以正常运行color detection例程吗？我使用esp32s3-eye开发板，但是烧录了官方例程后LCD板完全不亮，按键也没反应。

100312dog, 2024-08-19:
@Zhengqinze05 你好，阈值是hsv，h是一个环形，范围是0到180。这里的最小值156表示的是156-180, 最大值10表示的是0-10。

100312dog, 2024-08-19:
@Sean-WangXH 请问用的idf版本是什么，有报错信息嘛，建议使用idf 5.1。

addwsafreg, 2024-09-24:
{{Hmin, Hmax, Smin, Smax, Vmin, Vmax}, 64, "red"}, 阈值是这样设置的吗

> @Zhengqinze05 你好，阈值是hsv，h是一个环形，范围是0到180。这里的最小值156表示的是156-180, 最大值10表示的是0-10。

100312dog, 2024-09-24:
> {{Hmin, Hmax, Smin, Smax, Vmin, Vmax}, 64, "red"}, 阈值是这样设置的吗
> 
> > @Zhengqinze05 你好，阈值是hsv，h是一个环形，范围是0到180。这里的最小值156表示的是156-180, 最大值10表示的是0-10。

是的