espressif/esp-who #283: Esp-who face detection web compilation error

pixolish, 2024-01-13:
components/bus/8080_lcd_esp32s3.c:360:18: error: 'SOC_GDMA_PAIRS_PER_GROUP' undeclared (first use in this function); did you mean 'SOC_GDMA_PAIRS_PER_GROUP_MAX'?
  360 |     for (int x = SOC_GDMA_PAIRS_PER_GROUP - 1; x >= 0; x--) {
      |                  ^~~~~~~~~~~~~~~~~~~~~~~~
      |                  SOC_GDMA_PAIRS_PER_GROUP_MAX

Lzw655, 2024-01-15:
Hi @pixolish, could you please provide the version of ESP-IDF you are currently using? Also, are you using the examples from the esp-who repository? If so, please let me know which example you are using. This information will be helpful for reproducing the issue.

pixolish, 2024-01-17:
ESP-IDF v5.3-dev  - version of ESP-IDF
I am trying to build the examples:-
here is the path =  esp-who/examples/human_face_detection/web

PaulAnurag, 2024-10-17:
Hi, I have compiled it with ESP-IDF branch release5.0. Compilation is successfully done and also uploading of firmware but when I restart the board, I am not able to see anything on the terminal or the LCD.