espressif/esp-who #285: E (1078) spi: spicommon_bus_initialize_io(550): mosi not valid                        SPI BUS Initialization issue

Rahul14singh, 2024-02-13:
Board : Esp32-S2_Kaluga-1 v1.3
LCD : ESP-LyraP-LCD32 v1.2

Running example: esp-who\examples\human_face_detection\lcd

ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x1 (POWERON),boot:0x9 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3ffe6110,len:0x19c0
load:0x4004b000,len:0x4
load:0x4004b004,len:0xc34
load:0x4004f000,len:0x3428
entry 0x4004b1ec
I (23) boot: ESP-IDF v5.1.2 2nd stage bootloader
I (24) boot: compile time Feb 13 2024 22:01:11
I (24) boot: chip revision: v0.0
I (27) qio_mode: Enabling default flash chip QIO
I (32) boot.esp32s2: SPI Speed      : 80MHz
I (37) boot.esp32s2: SPI Mode       : QIO
I (42) boot.esp32s2: SPI Flash Size : 4MB
I (47) boot: Enabling RNG early entropy source...
I (52) boot: Partition Table:
I (56) boot: ## Label            Usage          Type ST Offset   Length
I (63) boot:  0 factory          factory app      00 00 00010000 003c0000
I (70) boot:  1 nvs              WiFi data        01 02 003d0000 00004000
I (78) boot:  2 fr               Unknown data     01 06 003e0000 00020000
I (85) boot: End of partition table
I (90) esp_image: segment 0: paddr=00010020 vaddr=3f000020 size=807c8h (526280) map
I (188) esp_image: segment 1: paddr=000907f0 vaddr=3ffc3720 size=02cech ( 11500) load
I (191) esp_image: segment 2: paddr=000934e4 vaddr=40024000 size=0cb34h ( 52020) load
I (206) esp_image: segment 3: paddr=000a0020 vaddr=40080020 size=69438h (431160) map
I (280) esp_image: segment 4: paddr=00109460 vaddr=40030b34 size=02be4h ( 11236) load
I (291) boot: Loaded app from partition at offset 0x10000
I (291) boot: Disabling RNG early entropy source...
I (302) cpu_start: Unicore app
I (302) cache: Instruction cache        : size 8KB, 4Ways, cache line size 32Byte
I (303) cache: Data cache               : size 8KB, 4Ways, cache line size 32Byte
I (310) esp_psram: Found 2MB PSRAM device
I (314) esp_psram: Speed: 40MHz
I (318) cpu_start: Pro cpu up.
I (732) esp_psram: SPI SRAM memory test OK
I (746) cpu_start: Pro cpu start user code
I (746) cpu_start: cpu freq: 160000000 Hz
I (746) cpu_start: Application information:
I (749) cpu_start: Project name:     human_face_detection_lcd
I (755) cpu_start: App version:      v0.9.4-181-gd3aa39b-dirty
I (762) cpu_start: Compile time:     Feb 13 2024 22:01:03
I (768) cpu_start: ELF file SHA256:  b7c9ce1b5428c574...
I (774) cpu_start: ESP-IDF:          v5.1.2
I (779) cpu_start: Min chip rev:     v0.0
I (784) cpu_start: Max chip rev:     v1.99
I (788) cpu_start: Chip rev:         v0.0
I (793) heap_init: Initializing. RAM available for dynamic allocation:
I (800) heap_init: At 3FFC9338 len 00032CC8 (203 KiB): DRAM
I (807) heap_init: At 3FFFC000 len 00003A10 (14 KiB): DRAM
I (813) heap_init: At 3FF9E000 len 00001FE8 (7 KiB): RTCRAM
I (819) esp_psram: Adding pool of 2048K of PSRAM memory to heap allocator
I (827) spi_flash: detected chip: generic
I (831) spi_flash: flash io: qio
I (842) app_start: Starting scheduler on CPU0
I (842) main_task: Started on CPU0
I (842) esp_psram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (852) main_task: Calling app_main()
I (852) who_camera: Camera module is ESP-S2-KALUGA
I (862) gpio: GPIO[2]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:2
I (872) cam_hal: cam init ok
I (872) sccb: pin_sda 8 pin_scl 7
I (872) sccb: sccb_i2c_port=1
I (892) camera: Detected camera at address=0x30
I (892) camera: Detected OV2640 camera
I (892) camera: Camera PID=0x26 VER=0x42 MIDL=0x7f MIDH=0xa2
I (972) s2 ll_cam: node_size: 3840, nodes_per_line: 1, lines_per_node: 8
I (972) s2 ll_cam: dma_half_buffer_min:  3840, dma_half_buffer: 11520, lines_per_half_buffer: 24, dma_buffer_size: 23040
I (982) cam_hal: buffer_size: 23040, half_buffer_size: 11520, node_buffer_size: 3840, node_cnt: 6, total_cnt: 10
I (992) cam_hal: Allocating 115200 Byte frame buffer in PSRAM
I (992) cam_hal: Allocating 115200 Byte frame buffer in PSRAM
I (1002) cam_hal: cam config ok
I (1002) ov2640: Set PLL: clk_2x: 1, clk_div: 3, pclk_auto: 1, pclk_div: 8
I (1082) who_lcd: Initialize SPI bus
E (1082) spi: spicommon_bus_initialize_io(550): mosi not valid
ESP_ERROR_CHECK failed: esp_err_t 0x102 (ESP_ERR_INVALID_ARG) at 0x40089abd
0x40089abd: register_lcd at C:/Espressif/frameworks/esp-idf-v5.1.2-2/workspace/esp-who/components/modules/lcd/who_lcd.c:52 (discriminator 1)

file: "/IDF/workspace/esp-who/components/modules/lcd/who_lcd.c" line 52
func: register_lcd
expression: spi_bus_initialize(SPI2_HOST, &bus_conf, SPI_DMA_CH_AUTO)

abort() was called at PC 0x4002b6bb on core 0
0x4002b6bb: _esp_error_check_failed at C:/Espressif/frameworks/esp-idf-v5.1.2-2/components/esp_system/esp_err.c:50

Backtrace: 0x40025642:0x3ffcd010 0x4002b6c5:0x3ffcd030 0x40031f56:0x3ffcd050 0x4002b6bb:0x3ffcd0c0 0x40089abd:0x3ffcd0f0 0x400895a9:0x3ffcd1a0 0x400e8ca3:0x3ffcd1c0 0x4002d669:0x3ffcd1f0
0x40025642: panic_abort at C:/Espressif/frameworks/esp-idf-v5.1.2-2/components/esp_system/panic.c:452
0x4002b6c5: esp_system_abort at C:/Espressif/frameworks/esp-idf-v5.1.2-2/components/esp_system/port/esp_system_chip.c:84
0x40031f56: abort at C:/Espressif/frameworks/esp-idf-v5.1.2-2/components/newlib/abort.c:38
0x4002b6bb: _esp_error_check_failed at C:/Espressif/frameworks/esp-idf-v5.1.2-2/components/esp_system/esp_err.c:50
0x40089abd: register_lcd at C:/Espressif/frameworks/esp-idf-v5.1.2-2/workspace/esp-who/components/modules/lcd/who_lcd.c:52 (discriminator 1)
0x400895a9: app_main at C:/Espressif/frameworks/esp-idf-v5.1.2-2/workspace/esp-who/examples/human_face_detection/lcd/main/app_main.cpp:15
0x400e8ca3: main_task at C:/Espressif/frameworks/esp-idf-v5.1.2-2/components/freertos/app_startup.c:208 (discriminator 13)
0x4002d669: vPortTaskWrapper at C:/Espressif/frameworks/esp-idf-v5.1.2-2/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:162

ELF file SHA256: b7c9ce1b5428c574

Rebooting...
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x3 (RTC_SW_SYS_RST),boot:0x9 (SPI_FAST_FLASH_BOOT)
Saved PC:0x400255b5
0x400255b5: esp_restart_noos_dig at C:/Espressif/frameworks/esp-idf-v5.1.2-2/components/esp_system/port/esp_system_chip.c:57 (discriminator 1)     

Am trying but not able to fix this issue. Tried changing PINs from:

#define BOARD_LCD_MOSI 47
#define BOARD_LCD_MISO -1
#define BOARD_LCD_SCK 21
#define BOARD_LCD_CS 44
#define BOARD_LCD_DC 43
#define BOARD_LCD_RST -1
#define BOARD_LCD_BL 48     

to 

#define LCD_CLK   GPIO_NUM_15
#define LCD_MOSI  GPIO_NUM_9
#define LCD_DC    GPIO_NUM_13
#define LCD_RST   GPIO_NUM_16
#define LCD_CS    GPIO_NUM_11
#define LCD_BK    GPIO_NUM_6  

But none of it worked. Need suggestions as to what may go wrong here. ESP PSRAM is enabled. Camera Config SC031GS is enabled.

Lzw655, 2024-02-20:
Hi @Rahul14singh, according to the [README](https://github.com/espressif/esp-who/tree/master/examples/human_face_detection#human-face-detection), this example is not valid for Esp32-S2_Kaluga-1 v1.3.

Rahul14singh, 2024-02-20:
@Lzw655 Thanks for pointing this out. I thought the terminal was working so others would work. Anyways Thanks :)

closed 2024-02-20 by Rahul14singh