espressif/esp-who #288: ESP32-S3-EYE cannot restore factory firmware

SamHall, 2024-03-29:
Following the [instructions ](https://github.com/espressif/esp-who/blob/master/docs/en/get-started/ESP32-S3-EYE_Getting_Started_Guide.md#17-default-firmware-and-function-test)and via various tools, I've not been able to successfully restore the features it shipped with. Instead, after each attempt, the board is no longer recognized as a USB device (Windows / USB connects/disconnects sounds in a loop). 

It's become a bit maddening. I come here hoping for some guidance. Thank you.

"Each ESP32-S3-EYE board comes with pre-built [default firmware](https://github.com/espressif/esp-who/tree/master/default_bin/esp32-s3-eye) that allows you to test its functions including voice wake-up, voice command recognition, face detection and recognition."

Sandra-lol, 2024-04-01:
Hi, have you tried flashing esp-who firmware? maybe that's why it keeps restarting.

SamHall, 2024-04-01:
> Hi, have you tried flashing esp-who firmware? maybe that's why it keeps restarting.

I've tried a lot of things, but don't recall that specifically. Are you saying that IF I had, it would have caused the reboot issue?  If so, wouldn't a reset and re-flash of other appropriate firmware correct the issue?

SamHall, 2024-04-01:
> Hi, have you tried flashing esp-who firmware? maybe that's why it keeps restarting.

FYI - the reason I used the firmware listed was based on the link from the Espressif site for this board.
[image]

Sandra-lol, 2024-04-02:
sorry, I meant lacking firmware caused the reboot...
Firmware here should be working: https://github.com/espressif/esp-who/tree/master/default_bin/esp32-s3-eye

SamHall, 2024-04-02:
> sorry, I meant lacking firmware caused the reboot... Firmware here should be working: https://github.com/espressif/esp-who/tree/master/default_bin/esp32-s3-eye

Yes, that was the link I referred to. Here's a screenshot of another firmware flash attempt (for Edge Impulse), showing the same behavior - that is, apparent success, but then can't read the device on the COM port. 
[image]

Sandra-lol, 2024-04-03:
Hi, we guess the there's hw issue with the board, maybe you can apply to change a board with the supplier?