espressif/esp-who #313: QR Code Scanner Example

iothub01, 2025-04-21:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

When i run Qr code example , it is showing camera image into LCD and not scanning qr code even when i display a qr code in front of it containing wifi information , it just show image in screen . do not scan it. 

### Actual behavior (suspected bug)

Instead it should be showing content of QR code 

### Error logs or terminal output

```shell

```

### Steps to reproduce the behavior

1. Run the code in ESP32P4 Dev board using esp idf v5.4.1

### Project release version

latest

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Windows

### Operating system version

window 10

### Shell

ZSH

100312dog, 2025-04-21:
@iothub01 The example not only display the result in lcd but also in terminal. If nothing shows in terminal, it means no qrcode is detected. Try moving the camera closer to the QR code.
I tested with the following qrcode, and the terminal output look like this.

[image]
[image]

iothub01, 2025-04-21:
I am getting this only - 
Board - ESP32P4

I (1145) ESPVideo: version: 0.8.0
I (1145) ESPVideo: driver:  MIPI-CSI
I (1146) ESPVideo: card:    MIPI-CSI
I (1146) ESPVideo: bus:     esp32p4:MIPI-CSI
I (1148) ESPVideo: capabilities:
I (1151) ESPVideo:      VIDEO_CAPTURE
I (1154) ESPVideo:      STREAMING
I (1157) ESPVideo: device capabilities:
I (1160) ESPVideo:      VIDEO_CAPTURE
I (1163) ESPVideo:      STREAMING
I (1219) main_task: Returned from app_main()

100312dog, 2025-04-21:
@iothub01 
Have no idea why it doesn't work if you use the esp32-p4-function-ev-board and this code.
Are the other examples in esp-who working? Human face detect or something else?

closed 2025-05-08 by iothub01