espressif/esp-who #314: 用esp32s3，配置lcd和摄像头和btn应该在哪里配置呀？

ccl-123, 2025-05-04:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

配置自定义摄像头、lcd、btn以实现human_face_recognition

### Actual behavior (suspected bug)

找不到配置的地方

### Error logs or terminal output

```shell

```

### Steps to reproduce the behavior

配置

### Project release version

最新

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Linux

### Operating system version

ubuntu22

### Shell

ZSH

100312dog, 2025-05-06:
@ccl-123
esp-who用了[bsp组件](https://components.espressif.com/components/espressif/esp32_s3_eye?language=en/versions/2.0.0)，外设的配置和开发板的型号强相关。如果你是自定义的板子，只是想改camera和lcd的 pin, 可以改这个
https://github.com/espressif/esp-bsp/blob/master/bsp/esp32_s3_eye/include/bsp/esp32_s3_eye.h#L59
button的配置在这个文件
https://github.com/espressif/esp-bsp/blob/master/bsp/esp32_s3_eye/esp32_s3_eye.c#L59

ccl-123, 2025-05-07:
在esp-who下找不到bsp组件呀，自定义的板子改了camera和lcd的 pin编译就能通过吗，别的地方有没有限制呢

ccl-123, 2025-05-07:
/home/cl/esp32_project/esp-who/components/who_recognition/who_recognition.cpp: In member function 'void who::recognition::WhoRecognition::create_btns()':
/home/cl/esp32_project/esp-who/components/who_recognition/who_recognition.cpp:146:31: error: cannot convert 'void (*)(void*, void*)' to 'button_event_args_t*'
  146 |         iot_button_register_cb(btns[1], BUTTON_SINGLE_CLICK, iot_btn_event_handler, (void *)m_btn_user_data));
      |         ~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                               |
      |                               void (*)(void*, void*)
/home/cl/esp32/esp-idf/components/esp_common/include/esp_err.h:116:30: note: in definition of macro 'ESP_ERROR_CHECK'
  116 |         esp_err_t err_rc_ = (x);                                        \
      |                              ^
In file included from /home/cl/esp32_project/esp-who/examples/human_face_recognition/managed_components/espressif__esp32_s3_eye_noglib/include/bsp/esp32_s3_eye.h:20,
                 from /home/cl/esp32_project/esp-who/examples/human_face_recognition/managed_components/espressif__esp32_s3_eye_noglib/include/bsp/esp-bsp.h:8,
                 from /home/cl/esp32_project/esp-who/components/who_peripherals/who_lcd/who_lcd.hpp:3,
                 from /home/cl/esp32_project/esp-who/components/who_frame_cap/who_frame_cap.hpp:3,
                 from /home/cl/esp32_project/esp-who/components/who_detect/who_detect_base.hpp:4,
                 from /home/cl/esp32_project/esp-who/components/who_detect/who_detect_lcd.hpp:2,
                 from /home/cl/esp32_project/esp-who/components/who_recognition/who_recognition.hpp:5:
/home/cl/esp32_project/esp-who/examples/human_face_recognition/managed_components/espressif__button/include/iot_button.h:107:105: note:   initializing argument 3 of 'esp_err_t iot_button_register_cb(button_handle_t, button_event_t, button_event_args_t*, button_cb_t, void*)'
  107 | esp_err_t iot_button_register_cb(button_handle_t btn_handle, button_event_t event, button_event_args_t *event_args, button_cb_t cb, void *usr_data);
      |                                                                                    ~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~
/home/cl/esp32_project/esp-who/components/who_recognition/who_recognition.cpp:149:31: error: cannot convert 'void (*)(void*, void*)' to 'button_event_args_t*'
  149 |         iot_button_register_cb(btns[3], BUTTON_SINGLE_CLICK, iot_btn_event_handler, (void *)(m_btn_user_data + 1)));
      |         ~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                               |
      |                               void (*)(void*, void*)
/home/cl/esp32/esp-idf/components/esp_common/include/esp_err.h:116:30: note: in definition of macro 'ESP_ERROR_CHECK'
  116 |         esp_err_t err_rc_ = (x);                                        \
      |                              ^
/home/cl/esp32_project/esp-who/examples/human_face_recognition/managed_components/espressif__button/include/iot_button.h:107:105: note:   initializing argument 3 of 'esp_err_t iot_button_register_cb(button_handle_t, button_event_t, button_event_args_t*, button_cb_t, void*)'
  107 | esp_err_t iot_button_register_cb(button_handle_t btn_handle, button_event_t event, button_event_args_t *event_args, button_cb_t cb, void *usr_data);
      |                                                                                    ~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~
/home/cl/esp32_project/esp-who/components/who_recognition/who_recognition.cpp:152:31: error: cannot convert 'void (*)(void*, void*)' to 'button_event_args_t*'
  152 |         iot_button_register_cb(btns[2], BUTTON_SINGLE_CLICK, iot_btn_event_handler, (void *)(m_btn_user_data + 2)));
      |         ~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                               |
      |                               void (*)(void*, void*)
/home/cl/esp32/esp-idf/components/esp_common/include/esp_err.h:116:30: note: in definition of macro 'ESP_ERROR_CHECK'
  116 |         esp_err_t err_rc_ = (x);                                        \
      |                              ^
/home/cl/esp32_project/esp-who/examples/human_face_recognition/managed_components/espressif__button/include/iot_button.h:107:105: note:   initializing argument 3 of 'esp_err_t iot_button_register_cb(button_handle_t, button_event_t, button_event_args_t*, button_cb_t, void*)'
  107 | esp_err_t iot_button_register_cb(button_handle_t btn_handle, button_event_t event, button_event_args_t *event_args, button_cb_t cb, void *usr_data);
      |                                                                                    ~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~
[1632/1819] Building CXX object...dir/who_recognition_app.cpp.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the /home/cl/esp32_project/esp-who/examples/human_face_recognition/build/log/idf_py_stderr_output_10549 and /home/cl/esp32_project/esp-who/examples/human_face_recognition/build/log/idf_py_stdout_output_10549

ccl-123, 2025-05-07:
human_face_recognition官方源例程编译失败

100312dog, 2025-05-08:
@ccl-123 修复拉，你可以再试一下。
https://components.espressif.com/components/espressif/button/versions/4.0.0/changelog?language=en
是因为button组件更新，api改了导致的编译失败。

ccl-123, 2025-05-08:
自定义的esp32s3和camera和lcd的 pin是要添加bsp组件之后在bsp组件里修改pin吗[https://github.com/espressif/esp-bsp/blob/master/bsp/esp32_s3_eye/include/bsp/esp32_s3_eye.h#L59](url)[](url)

100312dog, 2025-05-08:
@ccl-123 
两种方式。
一种你idf.py set-target 之后会在线拉取组件。在managed_components文件夹下面直接改拉取的组件代码。

https://docs.espressif.com/projects/idf-component-manager/en/latest/reference/manifest_file.html#override-path
第二种是用本地组件。下载组件源码到本地，然后在idf_component.yml里添加override_path。
https://github.com/espressif/esp-who/blob/master/components/who_peripherals/who_lcd/idf_component.yml#L8
https://github.com/espressif/esp-who/blob/master/components/who_peripherals/who_cam/idf_component.yml#L16

100312dog, 2025-05-08:
@ccl-123 
button是esp32_s3_eye_noglib的依赖，要改的话也是一样。
用本地组件也需要在esp32_s3_eye_noglib的idf_component.yml添加override_path。

ccl-123, 2025-05-08:
管脚在ESP32-S3-EYE-NOGLIB改还是bsp组件里面改，然后override_path覆盖就行了吗 @100312dog

100312dog, 2025-05-08:
ESP32-S3-EYE-NOGLIB就是bsp组件呀，
类似esp32_p4_function_ev_board_noglib也是。
就是一款板子会有一个自己的bsp组件。
你做的工作其实是修改ESP32-S3-EYE-NOGLIB，变成你自己custom板子的bsp组件。

ccl-123, 2025-05-08:
噢，非常感谢你！我去试一下

ccl-123, 2025-05-08:
@100312dog 噢还有一个问题就是摄像头和lcd的驱动问题，会不会不兼容，是st7789和ov2640吗

100312dog, 2025-05-08:
是st7789和ov2640

ccl-123, 2025-05-08:
@100312dog 
`const lvgl_port_display_cfg_t disp_cfg = {
        .io_handle = io_handle,
        .panel_handle = panel_handle,
        .buffer_size = BSP_LCD_DRAW_BUFF_SIZE,
        .double_buffer = BSP_LCD_DRAW_BUFF_DOUBLE,
        .hres = BSP_LCD_H_RES,
        .vres = BSP_LCD_V_RES,
        .monochrome = false,
        /* Rotation values must be same as used in esp_lcd for initial settings of the screen */
        .rotation =
            {
                .swap_xy = false,
                .mirror_x = false,
                .mirror_y = false,
            },
        .flags = {
            .buff_dma = true,
            .buff_spiram = false,
#if LVGL_VERSION_MAJOR >= 9
            .swap_bytes = false,
#endif
        }};`

 .buff_dma = true,
 .buff_spiram = false,官方例程这里没有使用psram，是否可以修改成使用psram分配，但是也会影响性能，有什么好的解决方法吗

ccl-123, 2025-05-08:
`const lvgl_port_display_cfg_t disp_cfg = {
        .io_handle = io_handle,
        .panel_handle = panel_handle,
        .buffer_size = BSP_LCD_DRAW_BUFF_SIZE,
        .double_buffer = BSP_LCD_DRAW_BUFF_DOUBLE,
        .hres = BSP_LCD_H_RES,
        .vres = BSP_LCD_V_RES,
        .monochrome = false,
        /* Rotation values must be same as used in esp_lcd for initial settings of the screen */
        .rotation =
            {
                .swap_xy = false,
                .mirror_x = false,
                .mirror_y = false,
            },
        .flags = {
            .buff_dma = true,
            .buff_spiram = false,
#if LVGL_VERSION_MAJOR >= 9
            .swap_bytes = false,
#endif
        }};`

官方例程esp-who/components/who_peripherals/who_lcd/who_lvgl_lcd.cpp这里 .buff_dma = true, .buff_spiram = false,没有使用psram，是否可以修改成使用psram分配，但是也会影响性能，有什么好的解决方法吗

100312dog, 2025-05-09:
@ccl-123 目前发布的esp-idf还没有添加s3 dma访问psram的支持。这个就算设置了.buff_spiram=true，最终也会在内部ram动态申请一块临时的空间，没有任何收益，并且有可能申请失败。所以目前s3只能用内部ram作为lcd buffer。
esp-idf 暂时不知道啥时候会合入这个支持。

ccl-123, 2025-05-09:
@100312dog 那为什么会分配失败呀，我把分辨率改成原来的240*240还是一样

[image]

100312dog, 2025-05-09:
@ccl-123 
你看下你用的esp_lvgl_port版本。如果是2.5.0之前的话有个bug
https://components.espressif.com/components/espressif/esp_lvgl_port/versions/2.6.0/changelog?language=en
[image]

如果不是版本的问题，可以尝试在这里加打印看看剩余多少内存，要申请多少内存
[image]

100312dog, 2025-05-09:
@ccl-123 这个可能不太方便呀，但是有问题还是可以提issue，会尽量回复

ccl-123, 2025-05-16:
@100312dog 
[image]
W (1039) spi_flash: Detected size(16384k) larger than the size in the binary image header(8192k). Using the size in the binary image header.
表示 ESP32-S3 芯片实际上拥有 16MB 的 Flash 存储，但固件（应用程序）是按照 8MB Flash 配置编译的。系统最终会以二进制镜像头中指定的较小尺寸（8MB）来操作 Flash，会导致问题吗？需要修改16mb吗，文件系统分区表这些要不要改，还是默认8MB就行

100312dog, 2025-05-16:
@ccl-123 要改sdkconfig， CONFIG_ESPTOOLPY_FLASHSIZE 改成16

ccl-123, 2025-05-16:
@100312dog 你好，我只修改esp32_s3_eye.h，
override_path: ``"components/esp32_s3_eye_noglib"之后烧录正常，但是
`I (1765) main_task: Returned from app_main()`这里是哪个地方可能出问题了呢。

100312dog, 2025-05-16:
@ccl-123 正常的就是这样呀。main只负责创建task。main结束了别的task还在运行。
如果lcd没显示，你可以在
https://github.com/espressif/esp-who/blob/master/components/who_detect/who_detect_base.cpp#L36
https://github.com/espressif/esp-who/blob/master/components/who_frame_cap/who_frame_cap.cpp#L33
加个打印，看看有没有正常不停地在循环。
有在循环多半还是外设相关的问题。

ccl-123, 2025-05-16:
@100312dog 是有在循环的，但确实是只修改了esp32_s3_eye.h

100312dog, 2025-05-16:
@ccl-123 你分辨率改的哪里？是这里吗？
https://github.com/espressif/esp-bsp/blob/master/bsp/esp32_s3_eye/include/bsp/display.h#L33

ccl-123, 2025-05-16:
@100312dog 
我没有改分辨率，240*240

[image]

100312dog, 2025-05-16:
@ccl-123 你之前不是说屏幕是240*320的嘛，换了一个？还是因为之前的报错，所以就没改。
你是每个示例都不行嘛，人脸检测也不行？

ccl-123, 2025-05-16:
我只弄了人脸识别，我屏幕是240*320，但是缓存不够，240*240刚好够，就改回官方的240*240了
然后我没上麦克风
[image]

ccl-123, 2025-05-16:
主要是lcd没亮，运行也没报什么错，没什么头绪

100312dog, 2025-05-16:
@ccl-123 麦克风没啥关系，所以还是lcd的问题。你可以试下人脸检测的例子，那个不会资源不够。分辨率改成正确的试下？

ccl-123, 2025-05-16:
嗯嗯我试试，还有我找不到这按键在哪配io口，是需要在别的文件吗

[image]

[image]

ccl-123, 2025-05-16:
@100312dog [https://github.com/espressif/esp-bsp/blob/5705ac7871989df3b9a84ece3ff8a44e0b5c1d63/bsp/esp32_s3_eye/include/bsp/esp32_s3_eye.h#L202](url)

[image]
这里这几个引脚都是NC,

ccl-123, 2025-05-16:
@100312dog 
这个就是esp32_s3_eye.h里的呀
[image]

ccl-123, 2025-05-16:
@100312dog 
人脸检测这个烧录后报错IIC冲突？

[image]

100312dog, 2025-05-16:
@ccl-123 你是不是又调了ESP_ERROR_CHECK(bsp_i2c_init());
又设置了pin_sccb_sda和pin_sccb_scl。
这俩设NC表示会直接用sccb_i2c_port设置的已经初始化的i2c。
所以BSP_CAMERA_DEFAULT_CONFIG不用改。
只需要改BSP_I2C_SCL，BSP_I2C_SDA，BSP_CAMERA*

ccl-123, 2025-05-16:
@100312dog 还真是，我去改一下，感谢

ccl-123, 2025-05-17:
@100312dog 我ov2640模块使用的是外部晶振固定12Mhz,需要改哪些吗？

ccl-123, 2025-05-17:
@100312dog 

[image]

[image]

pwdn没有配置引脚接哪里呢，接gnd?

100312dog, 2025-05-17:
@ccl-123 正常用的是ledc作为外部时钟，而不是晶振。
我看你之前人脸识别的例子，不是cam初始化都成功了吗？为什么人脸检测的又不行了？
pwdn，reset就是设nc阿

ccl-123, 2025-05-18:
@100312dog 我仔细看esp32_s3_eye原理图，里面的pwdn，reset接了GND和VDD,但是我按这样接还是不行；
我的ov2640是模块来的如图，模块的能实现吗？

[image]

ccl-123, 2025-05-18:
@100312dog 
能初始化，看门狗复位

[image]

100312dog, 2025-05-19:
@ccl-123 你之前人脸识别的例子，没有触发watchdog，并且一直在循环，说明相机没问题阿
这你之前贴的日志，明显已经执行到更后面了。
[image]
不需要外部时钟，或许xclk设nc?
camera驱动相关问题可以在[esp32-camera](https://github.com/espressif/esp32-camera)问

ccl-123, 2025-05-24:
@100312dog 
custom板没有ADC按键，能否定义三个普通按键实现对应的up、down、play功能，要修改哪些地方呢？