espressif/esp-who #316: 代码混乱，HAL、BSP、APP层代码杂揉到一起，建议重构。

crushonme, 2025-05-12:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Described the feature in detail and justified the reason for the request.
- [x] Provided specific use cases and examples.

### Feature description

ESP-WHO按道理是应用层，为什么在这里面包含了那么多BSP的逻辑？ 让第一次接触ESP 应用的人很抓狂。
正常来说移植到新的板子上，只用改BSP即可，而在这个项目里，如果不把完整逻辑看完，完全没法上手直接改。

### Use cases

举例说明： ESP-WHO默认支持的板子分别是ESP32-S3-EYE 和ESP32-P4-EYE。我的板子是微雪的ESP-S3-LCD-2，由于部分管脚配置不同，需要修改BSP来适配该项目。

正常而言只需要修改 ESP32_S3_EYE.H 里的配置即可，可实际并非如此。

在Application 层面竟然还看到如下本应该存在于BSP中的代码

[#if CONFIG_IDF_TARGET_ESP32P4
    auto cam = new WhoP4Cam(VIDEO_PIX_FMT_RGB565, 3, V4L2_MEMORY_USERPTR, true);
#elif CONFIG_IDF_TARGET_ESP32S3
    auto cam = new WhoS3Cam(PIXFORMAT_RGB565, FRAMESIZE_240X240, 2, true);
#endif](https://github.com/espressif/esp-who/blob/e7e6f23b19846fa910574136ef0178b3d715e2ba/examples/qrcode_recognition/main/app_main.cpp#L16)

100312dog, 2025-05-12:
@crushonme
正常而言只需要修改 ESP32_S3_EYE.H 里的配置即可，可实际并非如此。
所以如果只是管脚不一样，你还改了哪里呢？

#if CONFIG_IDF_TARGET_ESP32P4
auto cam = new WhoP4Cam(VIDEO_PIX_FMT_RGB565, 3, V4L2_MEMORY_USERPTR, true);
#elif CONFIG_IDF_TARGET_ESP32S3
auto cam = new WhoS3Cam(PIXFORMAT_RGB565, FRAMESIZE_240X240, 2, true);
#endif

这个是因为s3和p4 camera驱动的接口就有差异呀，没有办法写成一样的。你觉得有什么更好的写法，可以提出来呀。

crushonme, 2025-05-13:
例如在[camera_config_t camera_config = BSP_CAMERA_DEFAULT_CONFIG;](https://github.com/espressif/esp-who/blob/6d8b10961bdd70cbadcc80c0bcfb9fcb4d8217cf/components/who_peripherals/who_cam/who_cam_s3/who_s3_cam.cpp#L21)应用层传入分辨率等 default configuration，而default configuration中包含了引脚分配，这就很不合理。应用层传入的camera 参数应该是和板卡无关的。应用层应该配置分辨率、颜色空间等应用层配置，而具体的引脚配置应该通过HAL 隐藏在 Driver中。这样才能做到Application和Driver的解耦。

100312dog, 2025-05-14:
@crushonme
不懂你的意思。
现在代码应用层就是和其他的解耦的呀。
auto cam = new WhoS3Cam(PIXFORMAT_RGB565, FRAMESIZE_240X240, 2, true); 不就没有传入引脚。
就像你说的，现在改管脚也只需要修改bsp组件中的 [esp32_s3_eye.h](https://github.com/espressif/esp-bsp/blob/5705ac7871989df3b9a84ece3ff8a44e0b5c1d63/bsp/esp32_s3_eye/include/bsp/esp32_s3_eye.h#L72) 就好了呀。

至于[camera_config_t camera_config = BSP_CAMERA_DEFAULT_CONFIG;](https://github.com/espressif/esp-who/blob/6d8b10961bdd70cbadcc80c0bcfb9fcb4d8217cf/components/who_peripherals/who_cam/who_cam_s3/who_s3_cam.cpp#L21) 这么写也是因为其实使用了esp32-camera驱动的官方bsp里面都定义了这个BSP_CAMERA_DEFAULT_CONFIG。
[#define BSP_CAMERA_DEFAULT_CONFIG](https://github.com/espressif/esp-bsp/blob/5705ac7871989df3b9a84ece3ff8a44e0b5c1d63/bsp/esp32_s3_eye/include/bsp/esp32_s3_eye.h#L197)
[#define BSP_CAMERA_DEFAULT_CONFIG](https://github.com/espressif/esp-bsp/blob/5705ac7871989df3b9a84ece3ff8a44e0b5c1d63/bsp/esp32_s3_korvo_2/include/bsp/esp32_s3_korvo_2.h#L363)

如果你想要的是便捷地修改pin，感觉现在要看代码才能找到哪里改。
我可以把修改pin的地方暴露出来。
有两种方案: 
1. 增加sdkconfig
2. WhoS3Cam增加一个构造函数，包含可配置的入参camera_config_t

ccl-123, 2025-05-16:
@100312dog 增加sdkconfig吧，太多人都有这个custom板子的问题了，网上我没咋看到custom跑的例子，很多人都挺想玩一下这个组件的，但是引脚重定义太复杂了，改完了还是跑不起来

ccl-123, 2025-05-16:
@crushonme 你好，你的custom板子可以成功跑例程了吗？

crushonme, 2025-05-16:
> [@crushonme](https://github.com/crushonme) 不懂你的意思。 现在代码应用层就是和其他的解耦的呀。 auto cam = new WhoS3Cam(PIXFORMAT_RGB565, FRAMESIZE_240X240, 2, true); 不就没有传入引脚。 就像你说的，现在改管脚也只需要修改bsp组件中的 [...]

Okay. 大概明白你的意思了。所以你们的设计是通过头文件引入BSP_CAMERA_DEFAULT_CONFIG，然后应用层调用这个来初始化BSP。似乎也没毛病，不过从阅读代码来看，的确没有解耦。毕竟下面的代码是应用层，然后其中涉及了蛮多硬件层的差异。

[    camera_config_t camera_config = BSP_CAMERA_DEFAULT_CONFIG;
    camera_config.pixel_format = pixel_format;
    camera_config.frame_size = frame_size;
    camera_config.fb_count = fb_count;
    if (pixel_format == PIXFORMAT_JPEG) {
        camera_config.xclk_freq_hz = 20000000;
    }
    ESP_ERROR_CHECK(esp_camera_init(&camera_config));
    sensor_t *s = esp_camera_sensor_get();
    if (s->id.PID == OV3660_PID || s->id.PID == OV2640_PID || s->id.PID == GC032A_PID) {
        s->set_vflip(s, 1);
    }
    if (!horizontal_flip) {
        ESP_ERROR_CHECK(set_horizontal_flip());
    }](https://github.com/espressif/esp-who/blob/6d8b10961bdd70cbadcc80c0bcfb9fcb4d8217cf/components/who_peripherals/who_cam/who_cam_s3/who_s3_cam.cpp#L30)

crushonme, 2025-05-16:
> [@crushonme](https://github.com/crushonme) 你好，你的custom板子可以成功跑例程了吗？

跑起来了，不过初次接触这个，代码看的比较累。

ccl-123, 2025-05-17:
@crushonme 你好，请问具体一共要修改什么文件？能指教一下吗？

我改了esp32_s3_eye.h文件里的
BSP_I2C_SCL，BSP_I2C_SDA，BSP_CAMERA*和LCD引脚
（BSP_CAMERA_DEFAULT_CONFIG没改）
但是lcd一直没亮，烧录进去是这样

[image]

ccl-123, 2025-05-20:
@crushonme 你好
我看了下esp32s3-eye原理图，他lcd是12pin的，我的是8pin的也可以吗
还有我ov2640是模块用的外部晶振；
LCD_backlight_channel 中的.flags.output_invert 设置在哪个文件呀，谢谢