espressif/esp-who #317: esp32s3 使用espressif-quirc识别二维码，放到主函数中进行会不断重启

b1beyondmysely, 2025-05-14:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

我以为能够正常运行并将识别的二维码解码获取数据

### Actual behavior (suspected bug)

相反，将提取二维码quirc_extract(q, i, &code);和quirc_decode(&code, &data)加入进去后，导致跑到这就卡住，后续关机。将其注释掉，其他功能就能使用，能识别得到有二维码。
第一种情况就是  struct quirc_code code; struct quirc_data data;两个参数并没有给初始化值，也没用EXT_RAM_BSS_ATTR static定义到外部psram中，编译烧录进去开机，会不断重启 如图一

第二种是声明到外部psram中，如EXT_RAM_BSS_ATTR static struct quirc_code code ={0}; EXT_RAM_BSS_ATTR static struct quirc_data data={0}，编译烧录进去开机，跑到 quirc_extract提取二维码就卡住，后续关机如图二

[image]

[image]

### Error logs or terminal output

```shell
第一种得编译日志：
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2810,len:0x15a0
load:0x403c8700,len:0x4
load:0x403c8704,len:0xd24
load:0x403cb700,len:0x2f00
entry 0x403c8928
I (31) boot: ESP-IDF v5.4.1-361-g72de7111cc-dirty 2nd stage bootloader
I (31) boot: compile time May  9 2025 16:36:06
I (31) boot: Multicore bootloader
I (33) boot: chip revision: v0.2
I (36) boot: efuse block revision: v1.3
I (40) boot.esp32s3: Boot SPI Speed : 80MHz
I (44) boot.esp32s3: SPI Mode       : DIO
I (47) boot.esp32s3: SPI Flash Size : 4MB
I (51) boot: Enabling RNG early entropy source...
I (56) boot: Partition Table:
I (58) boot: ## Label            Usage          Type ST Offset   Length
I (65) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (71) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (78) boot:  2 factory          factory app      00 00 00010000 00100000
I (84) boot: End of partition table
I (87) esp_image: segment 0: paddr=00010020 vaddr=3c040020 size=127b0h ( 75696) map
I (108) esp_image: segment 1: paddr=000227d8 vaddr=3fc95d00 size=0d840h ( 55360) load
I (120) esp_image: segment 2: paddr=00030020 vaddr=42000020 size=33778h (210808) map
I (157) esp_image: segment 3: paddr=000637a0 vaddr=3fca3540 size=1bf3ch (114492) load
I (180) esp_image: segment 4: paddr=0007f6e4 vaddr=40374000 size=11bf8h ( 72696) load
I (196) esp_image: segment 5: paddr=000912e4 vaddr=600fe100 size=0001ch (    28) load
I (212) boot: Loaded app from partition at offset 0x10000
I (212) boot: Disabling RNG early entropy source...
I (222) esp_psram: Found 2MB PSRAM device
I (222) esp_psram: Speed: 40MHz
I (222) cpu_start: Multicore app
I (641) esp_psram: SPI SRAM memory test OK
I (668) cpu_start: Pro cpu start user code
I (668) cpu_start: cpu freq: 160000000 Hz
I (668) app_init: Application information:
I (668) app_init: Project name:     R128
I (672) app_init: App version:      1
I (675) app_init: Compile time:     May 13 2025 10:58:13
I (680) app_init: ELF file SHA256:  bc28c5930...
[... 752 lines cut ...]
        //     }

        //     if(connect_status)
        //     {
        //         Interface_Device_Connected();
        //     }
        //     else
        //     {
        //         Interface_Device_Disconnected();
        //     }

        //     if(Get_Voltage() < 3.4)
        //     {
        //         Interface_Low_Battery();
        //     }
        //     else
        //     {
        //         Clear_Area_Three();
        //     }

        //     status_bar_update_tick ++;
        // }
        //  vTaskDelay(200/portTICK_PERIOD_MS);
    }
}

100312dog, 2025-05-14:
@b1beyondmysely 可能是主函数stack不够。CONFIG_ESP_MAIN_TASK_STACK_SIZE 调大点，试试30000或更大

b1beyondmysely, 2025-05-14:
> [@b1beyondmysely](https://github.com/b1beyondmysely) 可能是主函数stack不够。CONFIG_ESP_MAIN_TASK_STACK_SIZE 调大点，试试30000或更大

我用命令打开menuconfig，选择component config，选择common ESP-related，进到里面发现没有选择0.0，该咋办

[image]

b1beyondmysely, 2025-05-14:
> [@b1beyondmysely](https://github.com/b1beyondmysely) 可能是主函数stack不够。CONFIG_ESP_MAIN_TASK_STACK_SIZE 调大点，试试30000或更大

给上25000往上就行了，非常感谢

closed 2025-05-14 by b1beyondmysely