espressif/esp-who #319: Can't detect face

via1968, 2025-05-22:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

在human face detect 0.2.3的情况下使用ov2640 直接拍摄rgb565的人脸时能够正常检测到人脸

### Actual behavior (suspected bug)

在human face detect 0.2.1的情况下一切正常，但是在human face detect 0.2.3的时候却无法检测到人脸

### Error logs or terminal output

```shell

```

### Steps to reproduce the behavior

进行人脸识别时

### Project release version

latest

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Linux

### Operating system version

N/A

### Shell

ZSH

### Additional context

extern "C" void app_main(void)
{

    ESP_ERROR_CHECK(fatfs_flash_mount());

    auto cam = new S3Cam(PIXFORMAT_RGB565, FRAMESIZE_240X240, 4, false);

    auto who_cam_lcd = new WhoCamLCD(cam);

    HumanFaceDetect *human_face_detect = new HumanFaceDetect();

    HumanFaceFeat *human_face_feat = new HumanFaceFeat();

    char db_path[64];
    snprintf(db_path, sizeof(db_path), "%s/face.db", CONFIG_SPIFLASH_MOUNT_POINT);

    auto human_face_recognizer =
        new HumanFaceRecognizer(human_face_feat, db_path);

    auto who_recognition = new WhoHumanFaceRecognition(human_face_detect, human_face_recognizer, cam);
    who_recognition->wifi_init();
    who_cam_lcd->run();
    who_recognition->run();
}

void WhoCamLCD::task(void *args)
{
    WhoCamLCD *self = (WhoCamLCD *)args;
    auto display_func_manager = DisplayFuncManager::get_instance();
    while (true) {

		//ESP_LOGI(TAG, "display");
		#ifdef FRAME_TEST
		camera_fb_t *fb = esp_camera_fb_get();

		esp_camera_fb_return(fb);
		#else
		uint8_t* camera_frame_temp_buf = ( uint8_t* )malloc( 240 * 240 * 2 );
		memcpy( camera_frame_temp_buf, camera_frame_buf, 240 * 240 * 2 );
		camera_frame9.buf = camera_frame_temp_buf;
		camera_fb_t *fb = &camera_frame9;
		self->m_cam->cam_fb_get();
        //self->m_cam->cam_fb_get1(fb);
		vTaskDelay(10/portTICK_PERIOD_MS);
        who::cam::cam_fb_t *fb1 = self->m_cam->cam_fb_peek(false);
		// printf("\r\n");
        // for(int i = 0; i < fb1->len; i++){
        // 		printf("%d\t", fb1->buf[i]);
        // }
		// printf("\r\n");
		display_func_manager->display(fb1);
		
        self->m_cam->cam_fb_return();
		free(camera_frame_temp_buf);
		camera_frame_temp_buf = NULL;
		
		#endif
    }
}

void WhoCamLCD::run()
{
	camera_frame_buf = ( uint8_t* )malloc( 240 * 240 * 2 );
	memcpy( camera_frame_buf, camera_frame9.buf, 240 * 240 * 2 );
    if (xTaskCreatePinnedToCore(task, "CamLCD", 4096, this, 5, nullptr, 0) != pdPASS) {
        ESP_LOGE(TAG, "Failed to create CamLCD task.\n");
    };
}

recognize_task

case fr_status_t::DETECT: {
            auto *fb = self->m_cam->cam_fb_peek();
            if (fb) {
                timestamp = fb->timestamp;
				auto img = who::cam::fb2img(fb);
                auto &det_res = self->m_detect->run(img);
                if (det_res.size() > 0) {
                    // ESP_LOGI("WhoDetect", "det_res.size :%d",det_res.size());
                    auto rec_res = self->m_recognizer->recognize(img, det_res);
                    // char *text = new char[64];
                    if (rec_res.empty()) {
                        
                        // strcpy(text, "who?");
						xSemaphoreTake(self->m_det_res_mutex, portMAX_DELAY);
						xSemaphoreTake(who::app::WhoHumanFaceRecognition::s_metadata_mutex, portMAX_DELAY);
						who::app::WhoHumanFaceRecognition::s_face_id[0] = 0;
						who::app::WhoHumanFaceRecognition::s_similarity[0] = 0;
						who::app::WhoHumanFaceRecognition::s_face_id[1] = 0;
						who::app::WhoHumanFaceRecognition::s_similarity[1] = 0;
						xSemaphoreGive(who::app::WhoHumanFaceRecognition::s_metadata_mutex);
						self->m_det_results.push({det_res, timestamp});
						xSemaphoreGive(self->m_det_res_mutex);
                        ESP_LOGI("WhoDetect", "who?");
                    } else {
						who::app::WhoHumanFaceRecognition::s_face_id[0] = 0;
						who::app::WhoHumanFaceRecognition::s_similarity[0] = 0;
						who::app::WhoHumanFaceRecognition::s_face_id[1] = 0;
						who::app::WhoHumanFaceRecognition::s_similarity[1] = 0;
                        for (size_t i = 0; i < rec_res.size(); i++) {
                            ESP_LOGI("WhoDetect",
                                     "rec_res.size :%d, id: %d, sim: %.2f",
                                     rec_res.size(),
                                     rec_res[i].id,
                                     rec_res[i].similarity);
                        }
						xSemaphoreTake(self->m_det_res_mutex, portMAX_DELAY);
						xSemaphoreTake(who::app::WhoHumanFaceRecognition::s_metadata_mutex, portMAX_DELAY);
						for(size_t i = 0; i < rec_res.size(); i++){
							who::app::WhoHumanFaceRecognition::s_face_id[i] = rec_res[i].id;
							who::app::WhoHumanFaceRecognition::s_similarity[i] = rec_res[i].similarity;
						}
						
						xSemaphoreGive(who::app::WhoHumanFaceRecognition::s_metadata_mutex);
						self->m_det_results.push({det_res, timestamp});
						xSemaphoreGive(self->m_det_res_mutex);
                        // snprintf(text, 64, "id: %d, sim: %.2f", rec_res[0].id, rec_res[0].similarity);
                    }
                    
                    
                } else {
                    //ESP_LOGI("WhoDetect", "no face");
					xSemaphoreTake(self->m_det_res_mutex, portMAX_DELAY);
					xSemaphoreTake(who::app::WhoHumanFaceRecognition::s_metadata_mutex, portMAX_DELAY);
					who::app::WhoHumanFaceRecognition::s_face_id[0] = -1;
					who::app::WhoHumanFaceRecognition::s_similarity[0] = -1;
					who::app::WhoHumanFaceRecognition::s_face_id[1] = -1;
					who::app::WhoHumanFaceRecognition::s_similarity[1] = -1;
					xSemaphoreGive(who::app::WhoHumanFaceRecognition::s_metadata_mutex);
                    self->m_det_results.push({det_res, timestamp});
                    xSemaphoreGive(self->m_det_res_mutex);
                }
            }

            break;
        }

via1968, 2025-05-22:
使用的是esp32s3n16r8

100312dog, 2025-05-22:
@via1968 如果esp-dl用的是在线的版本，确实应该会有问题，这周应该会重新传一下。

via1968, 2025-05-22:
好的，谢谢

closed 2025-05-22 by via1968