espressif/esp-who #324: Expand the api of esp-who to fully fledged API

mike88macedon, 2025-06-14:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Described the feature in detail and justified the reason for the request.
- [x] Provided specific use cases and examples.

### Feature description

Taking into consideration the following code :

auto model = new PedestrianDetect();
    auto detect = new WhoDetectAppLCD({{255, 0, 0}});
    detect->set_cam(cam);
    detect->set_lcd(lcd);
    detect->set_model(model);
    // detect->set_fps(5);
    detect->run();

We can only set the model cam,model,lcd and run the detect-->run task. But we are missing the functionallity of retrieving the detected results on the image.

### Use cases

Expand the API so we can get the detected results on the image it can be callback function that can return the detected results on an image frame. With this we will be able to do further processing with that data like sending it to a server or something like that. With the current implementation we are only limited to display the detected results on the image.

100312dog, 2025-06-16:
@mike88macedon 
https://github.com/espressif/esp-who/blob/master/components/who_detect/who_detect_base.hpp#L41
https://github.com/espressif/esp-who/blob/master/components/who_detect/who_detect_lcd.cpp#L26
https://github.com/espressif/esp-who/blob/master/components/who_detect/who_detect_term.cpp#L8

It's easy to achieve this by overriding the on_new_detect_result() function.
I can add something like set_new_detect_result_cb() in the next version.

mike88macedon, 2025-06-16:
> [@mike88macedon](https://github.com/mike88macedon) https://github.com/espressif/esp-who/blob/master/components/who_detect/who_detect_base.hpp#L41 [...]

Thanks that would be more convenient if we can implement the callback then deriving my own class that can display the results on the lcd and getting the detected metadata results on the on_new_detect_result().