espressif/esp-who #327: human_face_detect编译无法通过  error :'TaskStatus_t' {aka 'struct xTASK_STATUS'} has no member named 'xCoreID'

dxbsw, 2025-06-25:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

正常编译后下载到esp32S3eye，并成功运行

### Actual behavior (suspected bug)

'TaskStatus_t' {aka 'struct xTASK_STATUS'} has no member named 'xCoreID'

### Error logs or terminal output

```shell
D:/Project/4.ESP32S3_AI_CAN/esp-who/components/who_task/who_task_state.cpp: In member function 'void who::task::WhoTaskState::print_task_status()':     
D:/Project/4.ESP32S3_AI_CAN/esp-who/components/who_task/who_task_state.cpp:73:37: error: 'TaskStatus_t' {aka 'struct xTASK_STATUS'} has no member named 'xCoreID'
   73 |                task_status_array[i].xCoreID,
      |                                     ^~~~~~~
[1007/1247] Performing configure step for 'bootloader'
-- Found Git: D:/code/ESPRESSIF/ESP_IDF_TOOLS/V5.4.1/tools/idf-git/2.39.2/cmd/git.exe (found version "2.39.2.windows.1")
-- The C compiler identification is GNU 14.2.0
-- The CXX compiler identification is GNU 14.2.0
-- The ASM compiler identification is GNU
-- Found assembler: D:/code/ESPRESSIF/ESP_IDF_TOOLS/V5.4.1/tools/xtensa-esp-elf/esp-14.2.0_20241119/xtensa-esp-elf/bin/xtensa-esp32s3-elf-gcc.exe       
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: D:/code/ESPRESSIF/ESP_IDF_TOOLS/V5.4.1/tools/xtensa-esp-elf/esp-14.2.0_20241119/xtensa-esp-elf/bin/xtensa-esp32s3-elf-gcc.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: D:/code/ESPRESSIF/ESP_IDF_TOOLS/V5.4.1/tools/xtensa-esp-elf/esp-14.2.0_20241119/xtensa-esp-elf/bin/xtensa-esp32s3-elf-g++.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32s3
-- Project sdkconfig file D:/Project/4.ESP32S3_AI_CAN/esp-who/examples/human_face_detect/sdkconfig
-- Compiler supported targets: xtensa-esp-elf
-- Adding linker script D:/code/ESPRESSIF/ESP_IDF/V5.4.1/v5.4.1/esp-idf/components/soc/esp32s3/ld/esp32s3.peripherals.ld
-- Bootloader project name: "bootloader" version: 1
-- Adding linker script D:/code/ESPRESSIF/ESP_IDF/V5.4.1/v5.4.1/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.ld
-- Adding linker script D:/code/ESPRESSIF/ESP_IDF/V5.4.1/v5.4.1/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.api.ld
-- Adding linker script D:/code/ESPRESSIF/ESP_IDF/V5.4.1/v5.4.1/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.bt_funcs.ld
-- Adding linker script D:/code/ESPRESSIF/ESP_IDF/V5.4.1/v5.4.1/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.libgcc.ld
-- Adding linker script D:/code/ESPRESSIF/ESP_IDF/V5.4.1/v5.4.1/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.wdt.ld
-- Adding linker script D:/code/ESPRESSIF/ESP_IDF/V5.4.1/v5.4.1/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.version.ld
-- Adding linker script D:/code/ESPRESSIF/ESP_IDF/V5.4.1/v5.4.1/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.newlib.ld
-- Adding linker script D:/code/ESPRESSIF/ESP_IDF/V5.4.1/v5.4.1/esp-idf/components/bootloader/subproject/main/ld/esp32s3/bootloader.ld
-- Adding linker script D:/code/ESPRESSIF/ESP_IDF/V5.4.1/v5.4.1/esp-idf/components/bootloader/subproject/main/ld/esp32s3/bootloader.rom.ld
-- Components: bootloader bootloader_support efuse esp_app_format esp_bootloader_format esp_common esp_hw_support esp_rom esp_security esp_system esptool_py freertos hal log main micro-ecc newlib partition_table soc spi_flash xtensa
-- Component paths: D:/code/ESPRESSIF/ESP_IDF/V5.4.1/v5.4.1/esp-idf/components/bootloader D:/code/ESPRESSIF/ESP_IDF/V5.4.1/v5.4.1/esp-idf/components/bootloader_support D:/code/ESPRESSIF/ESP_IDF/V5.4.1/v5.4.1/esp-idf/components/efuse D:/code/ESPRESSIF/ESP_IDF/V5.4.1/v5.4.1/esp-idf/components/esp_app_format D:/code/ESPRESSIF/ESP_IDF/V5.4.1/v5.4.1/esp-idf/components/esp_bootloader_format D:/code/ESPRESSIF/ESP_IDF/V5.4.1/v5.4.1/esp-idf/components/esp_common D:/code/ESPRESSIF/ESP_IDF/V5.4.1/v5.4.1/esp-idf/components/esp_hw_support D:/code/ESPRESSIF/ESP_IDF/V5.4.1/v5.4.1/esp-idf/components/esp_rom D:/code/ESPRESSIF/ESP_IDF/V5.4.1/v5.4.1/esp-idf/components/esp_security D:/code/ESPRESSIF/ESP_IDF/V5.4.1/v5.4.1/esp-idf/components/esp_system D:/code/ESPRESSIF/ESP_IDF/V5.4.1/v5.4.1/esp-idf/components/esptool_py D:/code/ESPRESSIF/ESP_IDF/V5.4.1/v5.4.1/esp-idf/components/freertos D:/code/ESPRESSIF/ESP_IDF/V5.4.1/v5.4.1/esp-idf/components/hal D:/code/ESPRESSIF/ESP_IDF/V5.4.1/v5.4.1/esp-idf/components/log D:/code/ESPRESSIF/ESP_IDF/V5.4.1/v5.4.1/esp-idf/components/bootloader/subproject/main D:/code/ESPRESSIF/ESP_IDF/V5.4.1/v5.4.1/esp-idf/components/bootloader/subproject/components/micro-ecc D:/code/ESPRESSIF/ESP_IDF/V5.4.1/v5.4.1/esp-idf/components/newlib D:/code/ESPRESSIF/ESP_IDF/V5.4.1/v5.4.1/esp-idf/components/partition_table D:/code/ESPRESSIF/ESP_IDF/V5.4.1/v5.4.1/esp-idf/components/soc D:/code/ESPRESSIF/ESP_IDF/V5.4.1/v5.4.1/esp-idf/components/spi_flash D:/code/ESPRESSIF/ESP_IDF/V5.4.1/v5.4.1/esp-idf/components/xtensa
-- Configuring done (10.6s)
-- Generating done (0.4s)
-- Build files have been written to: D:/Project/4.ESP32S3_AI_CAN/esp-who/examples/human_face_detect/build/bootloader
[1021/1247] Building CXX object esp-idf/espressif__esp-dl/CMakeFiles/__idf_espressif__esp-dl.dir/vision/recognition/dl_feat_base.cpp.obj
ninja: build stopped: subcommand failed.
```

### Steps to reproduce the behavior

彻底清理后构建项目

### Project release version

最新的master 0eefcf9

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Windows

### Operating system version

win11

### Shell

ZSH

### Additional context

我拉取了最新的master，尝试使用human_face_detect，芯片选择ESP32S3，使用ESP32-S3-EYE开发板，编译无法通过，多次尝试结果一样，idf=5.4.1，是我还有什么地方没有配置好吗？

100312dog, 2025-06-30:
@dxbsw 看报错应该是没有选择到正确的sdkconfig文件

[image]
win下也需要设置这个环境变量。  

运行
```
idf.py reconfigure -DIDF_TARGET=esp32s3 -DSDKCONFIG_DEFAULTS=sdkconfig.bsp.esp32_s3_eye_noglib
```
命令行应该有类似这样的输出。这样说明环境变量正确配置，bsp_ext.py脚本生效了。
```
Executing action: set-bsp
Setting project for BSP: esp32_s3_eye_noglib
```

dxbsw, 2025-07-02:
我执行了
`idf.py reconfigure -DIDF_TARGET=esp32s3 -DSDKCONFIG_DEFAULTS=sdkconfig.bsp.esp32_s3_eye_noglib`
输出
`Executing action: .bsp.esp32_s3_eye_noglib
ninja: error: unknown target '.bsp.esp32_s3_eye_noglib'
command ".bsp.esp32_s3_eye_noglib" is not known to idf.py and is not a Ninja target`

还是不行，应该是esp32_s3_eye_noglib这个组件没有加载进来，我在文件中也没有找到
这个是现在的目录
[image]
 这个是b59241c6这个版本的目录
[image]

dxbsw, 2025-07-16:
@100312dog 
感谢，现在没有问题了，可以运行