espressif/esp-who #331: getting error. has no member xCoreID

iothub01, 2025-07-10:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

Run perfectly

### Actual behavior (suspected bug)

getting compilation error - 
esp-who/components/who_task/who_task_state.cpp:73:37: error: 'TaskStatus_t' {aka 'struct xTASK_STATUS'} has no member named 'xCoreID'
   73 |                task_status_array[i].xCoreID,

### Error logs or terminal output

```shell
esp-who/components/who_task/who_task_state.cpp:73:37: error: 'TaskStatus_t' {aka 'struct xTASK_STATUS'} has no member named 'xCoreID'
   73 |                task_status_array[i].xCoreID,
```

### Steps to reproduce the behavior

compile main branch 

### Project release version

latest

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Linux

### Operating system version

window 10 and using espressif extension in vscode

### Shell

ZSH

100312dog, 2025-07-10:
@iothub01 Duplicated with https://github.com/espressif/esp-who/issues/327
Have you follow the instructions of [quick-start](https://github.com/espressif/esp-who?tab=readme-ov-file#quick-start).
If you properly set the environment variable and still have this issue, please let me know.

100312dog, 2025-07-11:
@JasonYANG170 
I would like to help you solve the problem. There are some things I want to confirm.
Have you follow the instructions of [quick-start](https://github.com/espressif/esp-who?tab=readme-ov-file#quick-start)？
Is the environment variable IDF_EXTRA_ACTIONS_PATH properly set? 
What operating system are you using?

JasonYANG170, 2025-07-11:
> [@JasonYANG170](https://github.com/JasonYANG170) I would like to help you solve the problem. There are some things I want to confirm. Have you follow the instructions of [...]

I am using WSL2, ubuntu24.04, ubuntu20.04, IDF5.3\5.4\5.4 and all report errors

JasonYANG170, 2025-07-11:
Thank you, it works fine now. Maybe the previous sdkconfig was incorrectly configured, and I only deleted the build directory.

100312dog, 2025-07-14:
@JasonYANG170 
Try https://github.com/espressif/esp-who/tree/release/v1.1.0.
The problem you met may related to esp32-camera component version. 
Not sure whether release/v1.1.0 works on idfv5.0, but it should works on idfv5.1/idfv5.2/idfv5.3.

100312dog, 2025-07-30:
@iothub01 Please post the log after each command is executed.

100312dog, 2025-08-01:
@BB1989 运行 ```idf.py reconfigure```的日志有吗？

BB1989, 2025-08-01:
D:\esp-idf-v5.5\esp-who\examples\human_face_detect>idf.py reconfigure           
No module named 'click'
This usually means that "idf.py" was not spawned within an ESP-IDF shell environment or the python virtual environment used by "idf.py" is corrupted.        
Please use idf.py only in an ESP-IDF shell environment. If problem persists, please try to install ESP-IDF tools again as described in the Get Started guide.

BB1989, 2025-08-01:
[image]我感觉是不是需要通过菜单配置里面进行设置？  按照您刚才说的。是不是WHO开发不能用IDF+VSCODE的方式进行开发啊？

BB1989, 2025-08-01:
不能直接用VS 里面的那个编译按钮，只能通过命令行进行编译吗？

BB1989, 2025-08-01:
我感觉我好像在搭建环境的时候差了一步

BB1989, 2025-08-01:
我是不是必须要安装  ESP-IDF 工具安装器？  还是说直接在VSCODE 中通过

[image]安装就可以了？

100312dog, 2025-08-01:
@BB1989 对的，按钮只能完成一些简单的terminal等价操作

噢噢，好像是ctrl+shift+p，会有一个下拉窗出来。

[image]

或者下面有个按钮
[image]

BB1989, 2025-08-01:
-- Build files have been written to: D:/esp-idf-v5.5/esp-who/examples/human_face_detect/build
PS D:\esp-idf-v5.5\esp-who\examples\human_face_detect> idf.py reconfigure -DIDF_TARGET=target -DSDKCONFIG_DEFAULTS=sdkconfig.bsp.bsp_name
Executing action: reconfigure
Target 'target' specified on command line is not consistent with target 'esp32s3' in the environment.
PS D:\esp-idf-v5.5\esp-who\examples\human_face_detect>

100312dog, 2025-08-01:
@BB1989 

[image]

点这个按钮选择你芯片的型号

BB1989, 2025-08-01:
这个应该怎么添加啊。。不懂啊。。这两个分别是啥啊？不好意思啊。。一点儿不懂。现在芯片设置了啊。但是两个参数不知道写啥。。

100312dog, 2025-08-01:
@BB1989 不是呀，这边的reconfigure只是cmake了一下，后面才会build
target就是你板子芯片的型号，esp32s3或者esp32p4

[image]
每个例子下面会有一些sdkconfig开头的文件，如果你的板子是esp32s3eye，这个例子就是sdkconfig.bsp.esp32_s3_eye_noglib

执行
```
idf.py reconfigure -DIDF_TARGET="esp32s3" -DSDKCONFIG_DEFAULTS="sdkconfig.bsp.esp32_s3_eye_noglib"
```

BB1989, 2025-08-01:
> [@BB1989](https://github.com/BB1989) 不是呀，这边的reconfigure只是cmake了一下，后面才会build target就是你板子芯片的型号，esp32s3或者esp32p4
> 
> <img alt="Image" width="1315" height="934" src="https://private-user-images.githubusercontent.com/46221236/473360183-8fe207b0-8ffa-4e00-908e-e9e2b757d72a.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTQwMzc3NjMsIm5iZiI6MTc1NDAzNzQ2MywicGF0aCI6Ii80NjIyMTIzNi80NzMzNjAxODMtOGZlMjA3YjAtOGZmYS00ZTAwLTkwOGUtZTllMmI3NTdkNzJhLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTA4MDElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwODAxVDA4Mzc0M1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTg5YmQxM2ZjMjliMmVhOThmMjkzM2Q0N2RmOTA3YjhkNWU3NmM2NDgwMzU2Mjc1N2NkZmM1M2IxMTNhYjkzMzkmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.zypdnsmRLl00ZjYJ60PnpvEy1G0tKkG3m7lI23M6Xmc"> 每个例子下面会有一些sdkconfig开头的文件，如果你的板子是esp32s3eye，这个例子就是sdkconfig.bsp.esp32_s3_eye_noglib
> 执行
> 
> ```
> idf.py reconfigure -DIDF_TARGET="esp32s3" -DSDKCONFIG_DEFAULTS="sdkconfig.bsp.esp32_s3_eye_noglib"
> ```
执行完了,下面要执行啥呢？  idf.py build?

BB1989, 2025-08-01:
PS D:\esp-idf-v5.5\esp-who\examples\human_face_detect> & 'c:\Users\Cmm\.vscode\extensions\espressif.esp-idf-extension-1.10.2\export.ps1'
PS D:\esp-idf-v5.5\esp-who\examples\human_face_detect> idf.py reconfigure -DIDF_TARGET="esp32s3" -DSDKCONFIG_DEFAULTS="sdkconfig.bsp.esp32_s3_eye_noglib"
Executing action: set-bsp
Setting project for BSP: esp32_s3_eye_noglib
Executing action: reconfigure
Running cmake in directory D:\esp-idf-v5.5\esp-who\examples\human_face_detect\build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DPYTHON=D:\.esp-idf-v5.5\python_env\idf5.5_py3.11_env\Scripts\python.exe -DESP_PLATFORM=1 -DIDF_TARGET=esp32s3 -DSDKCONFIG_DEFAULTS=sdkconfig.bsp.esp32_s3_eye_noglib -DCCACHE_ENABLE=1 D:\esp-idf-v5.5\esp-who\examples\human_face_detect"...
-- Minimal build - OFF
-- ccache will be used for faster recompilation
-- Building ESP-IDF components for target esp32s3
NOTICE: Manifest files have changed, solving dependencies.
NOTICE: Skipping optional dependency: espressif/esp_video
..................NOTICE: Skipping optional dependency: espressif/esp_video
...NOTICE: Skipping optional dependency: espressif/esp_video
NOTICE: Updating lock file at D:\esp-idf-v5.5\esp-who\examples\human_face_detect\dependencies.lock.esp32s3
NOTICE: Processing 12 dependencies:
NOTICE: [1/12] espressif/button (4.1.3)
NOTICE: [2/12] espressif/cmake_utilities (0.5.3)
NOTICE: [3/12] espressif/esp-dl (3.1.5)
NOTICE: [4/12] espressif/esp32-camera (2.1.0)
NOTICE: [5/12] espressif/esp32_s3_eye_noglib (5.0.0~3)
NOTICE: [6/12] espressif/esp_codec_dev (1.3.6)
NOTICE: [7/12] espressif/esp_jpeg (1.3.1)     
NOTICE: [8/12] espressif/esp_new_jpeg (0.6.1) 
NOTICE: [9/12] espressif/human_face_detect (0.2.3)
NOTICE: [10/12] espressif/qma6100p (2.0.0)
NOTICE: [11/12] espressif/usb_host_uvc (2.3.0)
NOTICE: [12/12] idf (5.5.0)
NOTICE: Skipping optional dependency: espressif/esp_video
-- Project sdkconfig file D:/esp-idf-v5.5/esp-who/examples/human_face_detect/sdkconfig
warning: BSP_DISPLAY_LVGL_BUFFER_IN_PSRAM (defined at D:/esp-idf-v5.5/esp-who/examples/human_face_detect/managed_components/espressif__esp32_s3_eye_noglib/Kconfig:90) has leading or trailing whitespace in its prompt
Loading defaults file D:/esp-idf-v5.5/esp-who/examples/human_face_detect/sdkconfig.bsp.esp32_s3_eye_noglib...
-- Compiler supported targets: xtensa-esp-elf
-- USING O3
-- App "human_face_detect" version: v0.9.4-243-g31e3489-dirty
-- Adding linker script D:/esp-idf-v5.5/esp-who/examples/human_face_detect/build/esp-idf/esp_system/ld/memory.ld
-- Adding linker script D:/esp-idf-v5.5/esp-who/examples/human_face_detect/build/esp-idf/esp_system/ld/sections.ld.in
-- Adding linker script D:/esp-idf-v5.5/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.ld
-- Adding linker script D:/esp-idf-v5.5/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.api.ld
-- Adding linker script D:/esp-idf-v5.5/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.bt_funcs.ld
-- Adding linker script D:/esp-idf-v5.5/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.libgcc.ld
-- Adding linker script D:/esp-idf-v5.5/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.wdt.ld
-- Adding linker script D:/esp-idf-v5.5/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.version.ld
-- Adding linker script D:/esp-idf-v5.5/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.ble_master.ld
-- Adding linker script D:/esp-idf-v5.5/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.ble_50.ld
-- Adding linker script D:/esp-idf-v5.5/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.ble_smp.ld
-- Adding linker script D:/esp-idf-v5.5/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.ble_dtm.ld
-- Adding linker script D:/esp-idf-v5.5/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.ble_test.ld
-- Adding linker script D:/esp-idf-v5.5/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.ble_scan.ld
-- Adding linker script D:/esp-idf-v5.5/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.libc.ld
-- Adding linker script D:/esp-idf-v5.5/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.newlib.ld
-- Adding linker script D:/esp-idf-v5.5/esp-idf/components/soc/esp32s3/ld/esp32s3.peripherals.ld
-- BUTTON: 4.1.3
-- QMA6100P: 2.0.0
-- Components: app_trace app_update bootloader bootloader_support bt cmock console cxx driver efuse esp-tls esp_adc esp_app_format esp_bootloader_format esp_coex esp_common esp_driver_ana_cmpr esp_driver_bitscrambler esp_driver_cam esp_driver_dac esp_driver_gpio esp_driver_gptimer esp_driver_i2c esp_driver_i2s esp_driver_isp esp_driver_jpeg esp_driver_ledc esp_driver_mcpwm esp_driver_parlio esp_driver_pcnt esp_driver_ppa esp_driver_rmt esp_driver_sdio esp_driver_sdm esp_driver_sdmmc esp_driver_sdspi esp_driver_spi esp_driver_touch_sens esp_driver_tsens esp_driver_twai esp_driver_uart esp_driver_usb_serial_jtag esp_eth esp_event esp_gdbstub esp_hid esp_http_client esp_http_server esp_https_ota esp_https_server esp_hw_support esp_lcd esp_local_ctrl esp_mm esp_netif esp_netif_stack esp_partition esp_phy esp_pm esp_psram esp_ringbuf esp_rom esp_security esp_system esp_timer esp_vfs_console esp_wifi espcoredump 
espressif__button espressif__cmake_utilities espressif__esp-dl espressif__esp32-camera espressif__esp32_s3_eye_noglib espressif__esp_codec_dev espressif__esp_jpeg espressif__esp_new_jpeg espressif__human_face_detect espressif__qma6100p espressif__usb_host_uvc esptool_py fatfs freertos hal heap http_parser idf_test ieee802154 json log lwip main mbedtls mqtt newlib nvs_flash nvs_sec_provider openthrea [... cut ...]
      |         ^~~~~~~~~
[49/318] Building CXX object esp-idf/espressif__esp-dl/CMakeFiles/__idf_espressif__esp-dl.dir/dl/tensor/src/dl_tensor_base.cpp.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the D:\esp-idf-v5.5\esp-who\examples\human_face_detect\build\log\idf_py_stderr_output_1888 and D:\esp-idf-v5.5\esp-who\examples\human_face_detect\build\log\idf_py_stdout_output_1888
PS D:\esp-idf-v5.5\esp-who\examples\human_face_detect> 

还是报的一样的错。。

BB1989, 2025-08-01:
太感谢了，麻烦您了。可以了。总算编译过了。

Rotronica, 2026-09-09:
I'm getting this compilation error; I'm using ESP-IDF version v6.0.: [1081/1297] Building CXX object esp-idf/who_task/CMakeFiles/__idf_who_task.dir/who_task_state.cpp.obj
FAILED: esp-idf/who_task/CMakeFiles/__idf_who_task.dir/who_task_state.cpp.obj
C:\Espressif\tools\xtensa-esp-elf\esp-15.2.0_20251204\xtensa-esp-elf\bin\xtensa-esp32s3-elf-g++.exe -DESP_PLATFORM -DIDF_VER=\"v6.0\" -DSOC_MMU_PAGE_SIZE=CONFIG_MMU_PAGE_SIZE -DSOC_XTAL_FREQ_MHZ=CONFIG_XTAL_FREQ -D_GLIBCXX_HAVE_POSIX_SEMAPHORE -D_GLIBCXX_USE_POSIX_SEMAPHORE -D_GNU_SOURCE -D_POSIX_READER_WRITER_LOCKS -IE:/ESP32_PRUEBAS/esp-who/examples/human_face_recognition/build/config -IE:/ESP32_PRUEBAS/esp-who/components/who_task -IC:/esp/v6.0/esp-idf/components/esp_libc/platform_include -IC:/esp/v6.0/esp-idf/components/freertos/config/include -IC:/esp/v6.0/esp-idf/components/freertos/config/include/freertos -IC:/esp/v6.0/esp-idf/components/freertos/config/xtensa/include -IC:/esp/v6.0/esp-idf/components/freertos/FreeRTOS-Kernel/include -IC:/esp/v6.0/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include -IC:/esp/v6.0/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include/freertos -IC:/esp/v6.0/esp-idf/components/freertos/esp_additions/include -IC:/esp/v6.0/esp-idf/components/esp_hw_support/include -IC:/esp/v6.0/esp-idf/components/esp_hw_support/include/soc -IC:/esp/v6.0/esp-idf/components/esp_hw_support/ldo/include -IC:/esp/v6.0/esp-idf/components/esp_hw_support/debug_probe/include -IC:/esp/v6.0/esp-idf/components/esp_hw_support/etm/include -IC:/esp/v6.0/esp-idf/components/esp_hw_support/mspi_timing_tuning/include -IC:/esp/v6.0/esp-idf/components/esp_hw_support/mspi_timing_tuning/tuning_scheme_impl/include -IC:/esp/v6.0/esp-idf/components/esp_hw_support/power_supply/include -IC:/esp/v6.0/esp-idf/components/esp_hw_support/modem/include -IC:/esp/v6.0/esp-idf/components/esp_hw_support/include/soc/esp32s3 -IC:/esp/v6.0/esp-idf/components/esp_hw_support/port/esp32s3/. -IC:/esp/v6.0/esp-idf/components/esp_hw_support/port/esp32s3/include -IC:/esp/v6.0/esp-idf/components/esp_hw_support/mspi_timing_tuning/port/esp32s3/. -IC:/esp/v6.0/esp-idf/components/esp_hw_support/mspi_timing_tuning/port/esp32s3/include -IC:/esp/v6.0/esp-idf/components/heap/include -IC:/esp/v6.0/esp-idf/components/heap/tlsf -IC:/esp/v6.0/esp-idf/components/log/include -IC:/esp/v6.0/esp-idf/components/soc/include -IC:/esp/v6.0/esp-idf/components/soc/esp32s3 -IC:/esp/v6.0/esp-idf/components/soc/esp32s3/include -IC:/esp/v6.0/esp-idf/components/soc/esp32s3/register -IC:/esp/v6.0/esp-idf/components/hal/platform_port/include -IC:/esp/v6.0/esp-idf/components/hal/esp32s3/include -IC:/esp/v6.0/esp-idf/components/hal/include -IC:/esp/v6.0/esp-idf/components/esp_rom/include -IC:/esp/v6.0/esp-idf/components/esp_rom/esp32s3/include -IC:/esp/v6.0/esp-idf/components/esp_rom/esp32s3/include/esp32s3 -IC:/esp/v6.0/esp-idf/components/esp_rom/esp32s3 -IC:/esp/v6.0/esp-idf/components/esp_common/include -IC:/esp/v6.0/esp-idf/components/esp_system/include -IC:/esp/v6.0/esp-idf/components/esp_system/port/soc -IC:/esp/v6.0/esp-idf/components/esp_system/port/include/private -IC:/esp/v6.0/esp-idf/components/esp_stdio/include -IC:/esp/v6.0/esp-idf/components/xtensa/esp32s3/include -IC:/esp/v6.0/esp-idf/components/xtensa/include -IC:/esp/v6.0/esp-idf/components/xtensa/deprecated_include -IC:/esp/v6.0/esp-idf/components/esp_hal_gpio/include -IC:/esp/v6.0/esp-idf/components/esp_hal_gpio/esp32s3/include -IC:/esp/v6.0/esp-idf/components/esp_hal_usb/include -IC:/esp/v6.0/esp-idf/components/esp_hal_usb/esp32s3/include -IC:/esp/v6.0/esp-idf/components/esp_hal_pmu/include -IC:/esp/v6.0/esp-idf/components/esp_hal_pmu/esp32s3/include -IC:/esp/v6.0/esp-idf/components/esp_hal_ana_conv/include -IC:/esp/v6.0/esp-idf/components/esp_hal_ana_conv/esp32s3/include -IC:/esp/v6.0/esp-idf/components/esp_hal_dma/include -IC:/esp/v6.0/esp-idf/components/esp_hal_dma/esp32s3/include -IC:/esp/v6.0/esp-idf/components/lwip/include -IC:/esp/v6.0/esp-idf/components/lwip/include/apps -IC:/esp/v6.0/esp-idf/components/lwip/lwip/src/include -IC:/esp/v6.0/esp-idf/components/lwip/port/include -IC:/esp/v6.0/esp-idf/components/lwip/port/freertos/include -IC:/esp/v6.0/esp-idf/components/lwip/port/esp32xx/include -IC:/esp/v6.0/esp-idf/components/lwip/port/esp32xx/include/arch -IC:/esp/v6.0/esp-idf/components/lwip/port/esp32xx/include/sys @"E:/ESP32_PRUEBAS/esp-who/examples/human_face_recognition/build/toolchain/cxxflags" -fdiagnostics-color=always -fdiagnostics-color=always -ffunction-sections -fdata-sections -Wall -Werror -Wno-error=unused-function -Wno-error=unused-variable -Wno-error=unused-but-set-variable -Wno-error=deprecated-declarations -Wextra -Wno-error=extra -Wno-unused-parameter -Wno-sign-compare -Wno-enum-conversion -gdwarf-4 -ggdb -Og -fno-shrink-wrap -fmacro-prefix-map=E:/ESP32_PRUEBAS/esp-who/examples/human_face_recognition=. -fmacro-prefix-map=C:/esp/v6.0/esp-idf=/IDF -fstrict-volatile-bitfields -fno-jump-tables -fno-tree-switch-conversion -std=gnu++26 -fno-exceptions -fuse-cxa-atexit -MD -MT esp-idf/who_task/CMakeFiles/__idf_who_task.dir/who_task_state.cpp.obj -MF esp-idf\who_task\CMakeFiles\__idf_who_task.dir\who_task_state.cpp.obj.d -o esp-idf/who_task/CMakeFiles/__idf_who_task.dir/who_task_state.cpp.obj -c E:/ESP32_PRUEBAS/esp-who/components/who_task/who_task_state.cpp
E:/ESP32_PRUEBAS/esp-who/components/who_task/who_task_state.cpp: In member function 'void who::task::WhoTaskState::print_task_status()':
[... cut ...]
   73 |                task_status_array[i].xCoreID,
      |                                     ^~~~~~~
[1086/1297] Building CXX object esp-idf/esp_video_cxx/CMakeFiles/__idf_esp_video_cxx.dir/video_capture.cpp.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the E:\ESP32_PRUEBAS\esp-who\examples\human_face_recognition\build\log\idf_py_stderr_output_12188 and E:\ESP32_PRUEBAS\esp-who\examples\human_face_recognition\build\log\idf_py_stdout_output_12188